mcci-catena/arduino-lmic #884: How to set region without touching project_config/lmic_project_config.h

jpwsutton, 2022-07-07:
I'm building a pipeline that uses arduino-cli and the sketch.yaml project definition to build my code. However, I'm in europe and so need to use the LMIC_REGION_eu868 region. In the [readme](https://github.com/mcci-catena/arduino-lmic#selecting-the-lorawan-region-configuration) it states that you can use `-D CFG_eu868` build property to change it, but when you do that, the compiler gives the error: `# error You can define at most one of CFG_... variables`.

The arduino-cli argument I'm using is:
```
arduino-cli compile --profile lora_sensor --build-property "build.extra_flags=\"-D CFG_eu868\""
```

So it seems that the only way to correctly set the region would be to edit `project_config/lmic_project_config.h`, which isn't ideal in a pipeline situation.

Am I doing something obviously wrong here? Or is this an issue with the documentation / library?

**Environment**

This information is very important; it's hard to help without a complete set of answers.

- Version of LMIC being used. MCCI LoRaWAN LMIC library (4.1.1)
- Version of Arduino IDE being used: arduino-cli  Version: 0.24.0 Commit: c1b10f56 Date: 2022-06-21T14:33:24Z
- Network provider: Custom (multitech gateway)
- Region (EU868, US915, etc.)EU868
- Board: Adafruit Feather M0
- Radio HopeRF

terrillmoore, 2022-07-07:
Check how it's done in the CI scripts. More detailed answer later if I get a spare moment.

jpwsutton, 2022-07-07:
Ah, thanks. I've found what I think you were talking about:

```
-D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS
```

For reference to any others seeing a similar issue, it looks like the arduino-cli build.extra_flags property will override whatever is in your boards.txt [See this issue](https://github.com/arduino/arduino-cli/issues/210#issuecomment-493892258), so I've had to copy the flags from my boards.txt into the command so that I get everything all in one go:

```
arduino-cli compile --profile lora_sensor --build-property build.extra_flags="-DARDUINO_SAMD_ZERO -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0 -D__SAMD21G18A__ {build.usb_flags} -D CFG_eu868 -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS"
```

Thanks for your help!

closed 2022-07-07 by jpwsutton