mcci-catena/arduino-lmic #885: error: 'u1_t' does not name a type; did you mean 'uid_t'? (static const u1_t APPEUI[8] = { 0x86,....)

T365, 2022-07-18:
**Consider raising support questions on the forum first**

There is a discussion site, [forum.mcci.io](https://forum.mcci.io), which includes a [category focused on the Arduino LMIC](https://forum.mcci.io/c/device-software/arduino-lmic/5). Unless you're sure that your problem is a bug in the LMIC, it would be great if you can try for help there first. That will keep issues focused on work that needs to be done by the developers.

**Describe your question or issue**

Please give a clear and concise description of the problem you're facing and what you'd like help with.

**Environment**

This information is very important; it's hard to help without a complete set of answers.

- Version of LMIC being used. If using the latest github mainline, let us know, otherwise state the version.
- Version of Arduino IDE being used:
- Network provider (The Things Network, Swisscom, ChirpStack, etc.)
- Region (EU868, US915, etc.)
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.)
- Radio (HopeRF, SX1276, etc.)

T365, 2022-07-18:
LMIC: 4.2.0.1

After update,

Suddenly I got this message:
**error: 'u1_t' does not name a type; did you mean 'uid_t'?**

My code:
static const u1_t APPEUI[8] = { 0x86,......}

I tried to change u1_t to uid_t 
it compiles now, but my device will not connect now, I understand.

Can someone help me please ?

supcik, 2022-11-22:
This is perhaps linked to the issue #914 and fixed with the current master version. @T365, can you please try with the current version?