mcci-catena/arduino-lmic #887: Packet Queue

JoanaSTeixeira, 2022-07-20:
Hello,

I'm using the ttn-otaa example and the mcci-catena arduino-lmic library.
When I run the example I see the following in the terminal:

[image]

Why packet queue is only printed once?
Why is a new uplink sent without respecting TX_INTERNAL=60?

- Version of Arduino IDE being used:
- Network provider (ChirpStack)
- Region (EU868)
- Board (Arduino Uno v3.3)
- Radio (Dragino LoRa Shield)

terrillmoore, 2022-07-21:
The packet queued is printed by the setup() routine, which is only called once. After that, the flow is driven by completions. Meanwhile ChirpStack is (probably) sending a bunch of instructions as MAC commands after the join. My version of ChirpStack sets the packet rep count to 3 initially; but in any case, the LMIC will keep TXing as long as it's getting downlink MAC messages during the RX window. Based on the timing, you're getting downlink messages during RX1 (until the last one). You should look at the ChripStack logs to see more about what it's doing. Things look fine from here.

JoanaSTeixeira, 2022-07-21:
Thanks.
I changed the TX INTERVAL = 30, ran the program again, and found that it schedules a new transmission every 30 seconds.
How can I change in Chirpstack to set the packet rep count to 3 initially, for example?
From what I can see I have 4 right?

[image]

[image]