mcci-catena/arduino-lmic #888: Join in two differents network servers

JoanaSTeixeira, 2022-07-21:
Hi!

I want my device to join a network server, in order to eg 3 uplinks, unjoin this network.
In addition, I want it when it leaves a network to join another network server.

Assuming the keys to join both networks are predefined in the sckecth.

Is this possible to do?

I've seen these functions, but I don't know how to use them in the ttn-otaa example:
bit_t LMIC_startJoining(void);
void LMIC_tryRejoin(void);
void LMIC_unjoin(void);
void LMIC_unjoinAndRejoin(void);

- Version of Arduino IDE being used:
- Network provider (ChirpStack)
- Region (EU868)
- Board (Arduino Uno v3.3)
- Radio (Dragino LoRa Shield)

terrillmoore, 2022-07-22:
It's possible but not without a lot of work, sorry! It's very much an advanced topic. Perhaps someone else who's watching this may be able to help.

terrillmoore, 2022-07-22:
Converting to discussion.