mcci-catena/arduino-lmic #893: LMIC.snr and LMIC.rssi

JoanaSTeixeira, 2022-07-26:
Hi!

I want to print the SNR and RSSI received on the device.
I did it:

[image]

[image]

However, I'm not understanding what values are these that are printed in the terminal. Can you help me?

- Version of Arduino IDE being used:
- Network provider (ChirpStack)
- Region (EU868)
- Board (Arduino Uno v3.3)
- Radio (Dragino LoRa Shield)

JoanaSTeixeira, 2022-07-28:
How can I print in terminal SNR and RSSI dB values?

slavendam, 2023-10-24:
Change this
```
if (LMIC.txrxFlags & TXRX_ACK) {
                Serial.println("Received ack");
```
with this:
```
if (LMIC.txrxFlags & TXRX_ACK) {
                Serial.println("Received ack");
                Serial.println(LMIC.rssi);
                Serial.println(LMIC.snr);
}
```