mcci-catena/arduino-lmic #905: Extend the valid values read from regVersion in radio.c

drucknase, 2022-09-21:
from a 'stand alone' SX1276 the value read from regVersion will be 0x12 or from SX1272 the value will be 0x22. This values are verified in radio.c at line 1112 respectively at line 1115. If none of this two vales fit, the os_init() will return an error.
But there is at least a second value 0x13, which will be read from SX1276 regVersion, if the SX1276 is integrated in a SoC like the Acsip S76G. 
Would it be possible to change line 1112 so that value 0x12 and 0x13 will work with the LMIC library? 
My workaround for the moment is to edit radio.c . But with a next version of LMIC lib, I will lost the change and have to re-edit.
Thank you for your support and the immense work you do for LMIC library

terrillmoore, 2022-09-21:
Yes, as long as we have a reference for why we're using values of `regVersion`, I'm OK with adding them. I'll make this a priority for a point release... but I'm really suddenly extremely busy with MCCI's USB business, so it may be a while.

drucknase, 2022-09-21:
Thank you for your fast response. For me it's totaly OK what you say. I'm so happy, that I found this missmatch after days of frustration. So for now I can work with the workaround and it would be nice for me, and I believe others, when this will be realised in one of the next versions. Thanks again.