mcci-catena/arduino-lmic #911: LoRaWAN Regional Spec 1.0.3 Reference versions

Beetix, 2022-10-25:
Hi,

In the project's [README.md](https://github.com/mcci-catena/arduino-lmic#selecting-the-lorawan-region-configuration), LoRaWAN Regional Spec 1.0.3 Reference versions (like 2.2) are specified. How do they relate to the LoRa Alliance versions (like RP002-1.0.3)?

Thanks

terrillmoore, 2022-10-25:
Sorry, I don't know. I captured the relevant links at the time of doing the testing, but the LoRa Alliance may have changed things around; I have no control over what they do. If researching, I would use the Internet Archives/Wayback to find any broken links. Please submit a PR if you figure this out and find updated links.

Beetix, 2022-10-28:
Ok, I have looked at [https://web.archive.org/web/20211202134700/https://resources.lora-alliance.org/technical-specifications](https://web.archive.org/web/20211202134700/https://resources.lora-alliance.org/technical-specifications)