mcci-catena/arduino-lmic #912: Too many TX

Beetix, 2022-10-26:
**Describe the bug**

The device sends too many uplinks (at a 5s interval) at the beginning of the execution.

**Environment**

- LMIC: 4.1.1
- PlatformIO Core: 6.1.4
- Network provider: Chipstack
- Region: EU868
- Board : M5Stack Core 2 with [LoRa 868 module](https://github.com/Bjoerns-TB/M5Stack-LoRa-868-Network-Tester#installation-and-configuration)
- CPU: ESP32
- Radio: SX1276

**To Reproduce**

plaformio.ini
```
[env:m5stack-core2]
platform = espressif32
board = m5stack-core2
framework = arduino
lib_deps =
    m5stack/M5Core2@^0.1.2
    mcci-catena/MCCI LoRaWAN LMIC library@^4.1.1
monitor_speed = 115200
build_flags = 
	-D hal_init=LMIC_hal_init
	-D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS
	-D CFG_eu868
	-D CFG_sx1276_radio=1
```

main.cpp
```c
/*******************************************************************************
 * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
 * Copyright (c) 2018 Terry Moore, MCCI
 *
 * Permission is hereby granted, free of charge, to anyone
 * obtaining a copy of this document and accompanying files,
 * to do whatever they want with them without any restriction,
 * including, but not limited to, copying, modification and redistribution.
 * NO WARRANTY OF ANY KIND IS PROVIDED.
 *
 * This example sends a valid LoRaWAN packet with payload "Hello,
 * world!", using frequency and encryption settings matching those of
 * the The Things Network.
 *
 * This uses OTAA (Over-the-air activation), where where a DevEUI and
 * application key is configured, which are used in an over-the-air
 * activation procedure where a DevAddr and session keys are
 * assigned/generated for use with all further communication.
 *
 * Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in
 * g1, 0.1% in g2), but not the TTN fair usage policy (which is probably
 * violated by this sketch when left running for longer)!

 * To use this sketch, first register your application and device with
 * the things network, to set or generate an AppEUI, DevEUI and AppKey.
 * Multiple devices can use the same AppEUI, but each device has its own
 * DevEUI and AppKey.
 *
 * Do not forget to define the radio type correctly in
 * arduino-lmic/project_config/lmic_project_config.h or from your BOARDS.txt.
 *
 *******************************************************************************/
#include <M5Core2.h>
#include <lmic.h>
#include <hal/hal.h>
[... 205 lines cut ...]
615440: EV_JOINED
netid: 0
devaddr: 9B643E
AppSKey: 3B-C6-8B-E6-5B-FF-17-CD-04-59-65-C4-FA-52-DD-2D
NwkSKey: 15-6F-CA-81-C8-AA-45-3F-10-82-1E-D2-1A-5F-DD-1E
616191: EV_TXSTART
686273: EV_TXCOMPLETE (includes waiting for RX windows)
Next transmission = 4436280
1033155: EV_TXSTART
1101937: EV_TXCOMPLETE (includes waiting for RX windows)
Next transmission = 4851943
1322132: EV_TXSTART
1390914: EV_TXCOMPLETE (includes waiting for RX windows)
Next transmission = 5140921
1611107: EV_TXSTART
1679890: EV_TXCOMPLETE (includes waiting for RX windows)
Next transmission = 5429896
1900083: EV_TXSTART
2042310: EV_TXCOMPLETE (includes waiting for RX windows)
Next transmission = 5792316
5792338: EV_TXSTART
Packet queued
5935853: EV_TXCOMPLETE (includes waiting for RX windows)
Next transmission = 9685859
```

terrillmoore, 2022-10-26:
Sorry you're having problems.

I think you'll find that chirpstack is sending MAC messages to your device that require response. Each uplink from the device causes another downlink from Chirpstack, until Chirpstack has finished sending the MAC messages. I don't believe this is a bug in the LMIC, but rather known behavior with Chirpstack.

cyberman54, 2022-11-26:
I can confirm this, saw the same on a chirpstack network. The TX's are initiated by MAC commands from the network.

kadirozdinc, 2023-05-18:
I have a same problem with lmic library but when I use arduino-lorawan library instead of arduino-lmic library, I did not encounter this to many tx problem. What should we edit/add in this library to sort the problem out ?

Andermutu, 2024-05-31:
I found a solution, you have to disable MAC commands on chirpstack:

  # Mac-commands disabled.
  mac_commands_disabled=true