mcci-catena/arduino-lmic #922: PICO rp2040 not able to read serial1 while LMIC is active

SnkMtn000, 2023-02-28:
**Describe your question or issue**  GPS is not being read on Serial1 when using the lmic and hal libraries.  Is something blocking serial1?

Please give a clear and concise description of the problem you're facing and what you'd like help with.

Hooked up a Oled, Adafruit HopeRF 95 915 MHz on SPI0, and TinyGPSPlus neo6 on Serial1 on a bread board using code from the example ttn-otta.ino.

Code works great with my breadboard wiring if only reading GPS module.  Adding the Lorawan code it fails to read serial1 with the addition of the LMIC code.  Despite the GPS not being read it it shows up on the things network and delivers a dummy payload correctly.  

**Environment**
- Version of LMIC being used. 4.1.1
- Version of Arduino IDE being used: 2.04
- Network provider : The Things Network
- Region US915
- Board Raspberry Pi PICO rp2040
- Radio Adafruits HopeRF 95 break out board

/* 
  PICO, Oled, add LORA
*/

#include <Wire.h>
#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>
#include <Adafruit_SSD1306.h>
#include <TinyGPSPlus.h>

#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
#define OLED_RESET     -1 // Reset pin # (or -1 if sharing Arduino reset pin)
#define SCREEN_ADDRESS 0x3C ///< See datasheet for Address; 0x3C 

int readSensorDelay = 5000;
const unsigned TX_INTERVAL = 10;  // was 60 Schedule TX every this many seconds (might become longer due to duty
static const u1_t PROGMEM APPEUI[8]={ 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };  //  LSB
void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}
static const u1_t PROGMEM DEVEUI[8]={ Nope! };  // LSB
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}
static const u1_t PROGMEM APPKEY[16] = { Nope! };  //  MSB
void os_getDevKey (u1_t* buf) {  memcpy_P(buf, APPKEY, 16);}
static osjob_t sendjob;
const lmic_pinmap lmic_pins = {  // Pin mapping  Feather MO
    .nss = 17,  // was 6  was 8
    .rxtx = LMIC_UNUSED_PIN,  //  was LMIC_UNUSED_PIN
    .rst = 22,  //  was 5
    .dio = {20, 21, LMIC_UNUSED_PIN},  //   was 2 3
};
int count = 0;
int mycount = 0;
int myhr, mymin, mysec;
int mysat = 0;
float mylat = 0.0;
float mylon = 0.0;
float mytemp = 0.00;
String myString = "";

//uint8_t mydata[] = "012345678901234567890123456789012345678901234567890123456789";  // = 60 characters
uint8_t mydata[] = "0123456789012345678901234567890123456789012345678901234567890123";  // = 64 characters

//uint8_t mydata[] = "**************************************************";  // = 50 characters
int myintdata[80];

TinyGPSPlus gps;
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

void setup()
{
  Serial.begin(115200);
  Serial1.begin(9600);
  initOled();
[... 273 lines cut ...]
  Serial.print("SCK = ");Serial.println(SCK);
  Serial.print("CS = ");Serial.println(SS);
  Serial.print("lmic NSS = "); Serial.println(lmic_pins.nss);
  Serial.print("lmic rst = "); Serial.println(lmic_pins.rst);
  Serial.print("lmic SPI FREQ = "); Serial.println(lmic_pins.spi_freq);
  Serial.print("lmic DIO 0 = "); Serial.println(lmic_pins.dio[0]);
  Serial.print("lmic DIO 1 = "); Serial.println(lmic_pins.dio[1]);
  //Serial.print("lmic pConfig = "); Serial.println(lmic_pins.pConfig);
  Serial.print("lmic rssi cal = "); Serial.println(lmic_pins.rssi_cal);
  Serial.print("lmic rxtx = "); Serial.println(lmic_pins.rxtx);
  Serial.print("lmic rxtx rx active = "); Serial.println(lmic_pins.rxtx_rx_active);
  Serial.print("Satellites = "); Serial.println(mysat);
  Serial.print("Latitude = ");Serial.println(mylat);
  Serial.print("Longitude = "); Serial.println(mylon);
  Serial.println();
  delay(1000);

}

void printHex2(unsigned v) {
    v &= 0xff;
    if (v < 16)
        Serial.print('0');
    Serial.print(v, HEX);
}

terrillmoore, 2023-03-01:
Added markdown code fence for readability.

terrillmoore, 2023-03-01:
HI @SnkMtn000, sorry you're having problems.

I don't have a pico, and am on the road at the moment, so this is a theoretical response.

Your description sounds like a problem with SPI pins and GPIOs conflicting with the pins used for `Serial1`.  It could be due to the board support package -- there's a lot of "automatic" stuff that can cause confusion.

I would try calling the LMIC initialization code and *not* transmit data, see if that blocks `Serial1` operation.

If it does, I'd go through the LMIC init code and copy pieces to the test sketch, until I find the thing that breaks `Serial1`.

If it doesn't, I'd document exactly what has to be done to break `Serial1` (what LMIC routines need to be called), again using a simplified, small test sketch.

Best regards,
--Terry

SnkMtn000, 2023-03-06:
Was able to figure it out.  Thanks for your suggestions.  Turns out code is working as designed.  My GPS battery backup was very discharged causing a very long wait for an initial satellite lock.

closed 2023-03-06 by SnkMtn000