mcci-catena/arduino-lmic #931: ESP32 board not supported

laurentvolpon777, 2023-04-25:
Hello,
I'm trying to compile the ttn-otaa example with my ESP32 but I have this...
c:\Users\...\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\hal\getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
c:\Users\...\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\hal\getpinmap_thisboard.cpp:71:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")

I thought ESP32 was compatible with this library.
Thanks for your help
Laurent

cyberman54, 2023-05-07:
if you do what the hint suggests, providing an explicit pinmap to the lib, it will work with ESP32 boards.

dadanugm, 2023-06-29:
Hi @laurentvolpon777 
Have you resolved the issue? this happen also for Board Nucleo 64. This unsupported board turn into oslmic init failure.
[image]
Any idea to solve?
Thanks

laurentvolpon777, 2023-06-29:
@cyberman54 Sorry, I missed your answer. I'm a newbie and not sure how to proceed. Could you please help me on this.

@dadanugm I didn't resolve my problem. I tried with another microcontroller (Seeed XIAO - LoRa E5), but I would prefer to go back with my dear ESP32.