mcci-catena/arduino-lmic #938: How to make DR fixed in OTAA example

kadirozdinc, 2023-05-31:
Hello, I implement ttn_otaa example successfully and I need to make DR fixed at SF12. I add these but doesn't work :  

    LMIC_setAdrMode(false);
    LMIC_setDrTxpow(DR_SF12, 14);

Codes are below : 

```
#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

//
// For normal use, we require that you edit the sketch to replace FILLMEIN
// with values assigned by the TTN console. However, for regression tests,
// we want to be able to compile these scripts. The regression tests define
// COMPILE_REGRESSION_TEST, and in that case we define FILLMEIN to a non-
// working but innocuous value.
//
#ifdef COMPILE_REGRESSION_TEST
# define FILLMEIN 0
#else
# warning "You must replace the values marked FILLMEIN with real values from the TTN control panel!"
# define FILLMEIN (#dont edit this, edit the lines that use FILLMEIN)
#endif

// This EUI must be in little-endian format, so least-significant-byte
// first. When copying an EUI from ttnctl output, this means to reverse
// the bytes. For TTN issued EUIs the last bytes should be 0xD5, 0xB3,
// 0x70.
static const u1_t PROGMEM APPEUI[8]={ FILLMEIN };
void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}

// This should also be in little endian format, see above.
static const u1_t PROGMEM DEVEUI[8]={ FILLMEIN };
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}

// This key should be in big endian format (or, since it is not really a
// number but a block of memory, endianness does not really apply). In
// practice, a key taken from ttnctl can be copied as-is.
static const u1_t PROGMEM APPKEY[16] = { FILLMEIN };
void os_getDevKey (u1_t* buf) {  memcpy_P(buf, APPKEY, 16);}

static uint8_t mydata[] = "Hello, world!";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 60;

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 6,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 5,
    .dio = {2, 3, 4},
};

void printHex2(unsigned v) {
    v &= 0xff;
    if (v < 16)
        Serial.print('0');
    Serial.print(v, HEX);
}

void onEvent (ev_t ev) {
    Serial.print(os_getTime());
    Serial.print(": ");
    switch(ev) {
        case EV_SCAN_TIMEOUT:
            Serial.println(F("EV_SCAN_TIMEOUT"));
            break;
        case EV_BEACON_FOUND:
            Serial.println(F("EV_BEACON_FOUND"));
            break;
        case EV_BEACON_MISSED:
            Serial.println(F("EV_BEACON_MISSED"));
            break;
        case EV_BEACON_TRACKED:
            Serial.println(F("EV_BEACON_TRACKED"));
            break;
        case EV_JOINING:
            Serial.println(F("EV_JOINING"));
            break;
        case EV_JOINED:
            Serial.println(F("EV_JOINED"));
            {
              u4_t netid = 0;
              devaddr_t devaddr = 0;
              u1_t nwkKey[16];
              u1_t artKey[16];
              LMIC_getSessionKeys(&netid, &devaddr, nwkKey, artKey);
              Serial.print("netid: ");
              Serial.println(netid, DEC);
              Serial.print("devaddr: ");
              Serial.println(devaddr, HEX);
              Serial.print("AppSKey: ");
              for (size_t i=0; i<sizeof(artKey); ++i) {
                if (i != 0)
                  Serial.print("-");
                printHex2(artKey[i]);
              }
              Serial.println("");
              Serial.print("NwkSKey: ");
              for (size_t i=0; i<sizeof(nwkKey); ++i) {
                      if (i != 0)
                              Serial.print("-");
                      printHex2(nwkKey[i]);
              }
              Serial.println();
            }
            // Disable link check validation (automatically enabled
            // during join, but because slow data rates change max TX
	    // size, we don't use it in this example.
            LMIC_setLinkCheckMode(0);
            break;
        /*
        || This event is defined but not used in the code. No
        || point in wasting codespace on it.
        ||
        || case EV_RFU1:
        ||     Serial.println(F("EV_RFU1"));
        ||     break;
        */
        case EV_JOIN_FAILED:
            Serial.println(F("EV_JOIN_FAILED"));
            break;
        case EV_REJOIN_FAILED:
            Serial.println(F("EV_REJOIN_FAILED"));
            break;
        case EV_TXCOMPLETE:
            Serial.println(F("EV_TXCOMPLETE (includes waiting for RX windows)"));
            if (LMIC.txrxFlags & TXRX_ACK)
              Serial.println(F("Received ack"));
            if (LMIC.dataLen) {
              Serial.print(F("Received "));
              Serial.print(LMIC.dataLen);
              Serial.println(F(" bytes of payload"));
            }
            // Schedule next transmission
            os_setTimedCallback(&sendjob, os_getTime()+sec2osticks(TX_INTERVAL), do_send);
            break;
        case EV_LOST_TSYNC:
            Serial.println(F("EV_LOST_TSYNC"));
            break;
        case EV_RESET:
            Serial.println(F("EV_RESET"));
            break;
        case EV_RXCOMPLETE:
            // data received in ping slot
            Serial.println(F("EV_RXCOMPLETE"));
            break;
        case EV_LINK_DEAD:
            Serial.println(F("EV_LINK_DEAD"));
            break;
        case EV_LINK_ALIVE:
            Serial.println(F("EV_LINK_ALIVE"));
            break;
        /*
        || This event is defined but not used in the code. No
        || point in wasting codespace on it.
        ||
        || case EV_SCAN_FOUND:
        ||    Serial.println(F("EV_SCAN_FOUND"));
        ||    break;
        */
        case EV_TXSTART:
            Serial.println(F("EV_TXSTART"));
            break;
        case EV_TXCANCELED:
            Serial.println(F("EV_TXCANCELED"));
            break;
        case EV_RXSTART:
            /* do not print anything -- it wrecks timing */
            break;
        case EV_JOIN_TXCOMPLETE:
            Serial.println(F("EV_JOIN_TXCOMPLETE: no JoinAccept"));
            break;

        default:
            Serial.print(F("Unknown event: "));
            Serial.println((unsigned) ev);
            break;
    }
}

void do_send(osjob_t* j){
    // Check if there is not a current TX/RX job running
    if (LMIC.opmode & OP_TXRXPEND) {
        Serial.println(F("OP_TXRXPEND, not sending"));
    } else {
        // Prepare upstream data transmission at the next possible time.
        LMIC_setTxData2(1, mydata, sizeof(mydata)-1, 0);
        Serial.println(F("Packet queued"));
    }
    // Next TX is scheduled after TX_COMPLETE event.
}

void setup() {
    Serial.begin(9600);
    Serial.println(F("Starting"));

    #ifdef VCC_ENABLE
    // For Pinoccio Scout boards
    pinMode(VCC_ENABLE, OUTPUT);
    digitalWrite(VCC_ENABLE, HIGH);
    delay(1000);
    #endif

    // LMIC init
    os_init();
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();

    LMIC_setAdrMode(false);
    LMIC_setDrTxpow(DR_SF12, 14);

    // Start job (sending automatically starts OTAA too)
    do_send(&sendjob);
}

void loop() {
    os_runloop_once();
}
```

How can i implement it correctly ?

i-g-g-y, 2023-06-28:
I can confirm the same issue. The code bellow will not put a node into a fixed data rate.

`
#if STATIONARY_NODE
  //stationary node should use LoRaWAN ADR feature (ADR = adaptive data rate)
  //do not use LMIC_setDrTxpow() with ADR enabled
  #else
  //moving node should not use LoRaWAN ADR feature (ADR = adaptive data rate)
  LMIC_setAdrMode(0);
  LMIC_setDrTxpow(DR_SF10,14);
  #endif
`

miguemorales, 2023-07-14:
Hi, I am also using this library, and i can report the same issue, but i can go further. To me, it happens that the device does de join and the first message with SF7, but the next message are sent with the SF that you order. I don't know why it happens like that, but it really bother us, bucause usually the first join is the most difficult message, then the next messages do not have problems. If i power down the device, next join and first message again uses SF7, and after that it returns the specified SF.
Kind regards

kadirozdinc, 2023-07-14:
While digging around the library, I realized that this issue is related to Payload Size. If you look over "LoraWAN Regional Parameters" document, you will see Maximum payload size differs per DataRate. For example for SF12 and SF11 MaxPayloadSize is 51byte (FRMpayload). If in your example, you set DR0(SF12) and try to send payload that exceed this limitation, library decrease SF by itself. Please lower your payload below that limit and try again. For more info refer to Regional parameters documantation

i-g-g-y, 2023-07-16:
I can also confirm, that setting the DR rate just before sending packet out works, should be written in the README.

slavendam, 2023-10-24:
If you set up SF before join that parameter will be deleted during join.

In EU868 region (and AS923 I think also) device will try to join on SF7 and if that fails it will increase SF and repeat. Than it will try to join on SF8, and if it fails it will increase SF on SF9. It will repeat this until SF12 and than it will try only on SF12.
In US and AU region device is always trying to join on SF10.

If you want to set fixed SF, you need to do that in EV_JOINED part, next to "LMIC_setLinkCheckMode(0);". So after join is successful, than set up SF you want and disable ADR.
Note that if device joined on e.g. SF11 it means that it couldn't reach gateway with lower SFs. So if you after join put e.g. SF8 or SF9, it is possible that you device will not be able to transfer messages to gateway.

martinius96, 2023-10-30:
@slavendam 
Can you provide full example please?

slavendam, 2023-10-30:
> Can you provide full example please?

1. Press CTRL+F while on this page, 
2. find line "LMIC_setLinkCheckMode(0);"  in example above
3. next to that add:
```
LMIC_setAdrMode(false);
LMIC_setDrTxpow(DR_SF12, 14); // or other SF you want
```

martinius96, 2023-10-30:
I am using this example with ABP method, but I am facing same problem.
https://github.com/martinius96/hladinomer-studna-scripty/blob/master/examples/Hladinomer/LoRaWAN/LoRaWAN.ino

I tried your suggestion, I added to EV_JOINED case:
`LMIC_setLinkCheckMode(0);`
`LMIC_setAdrMode(false);`
`LMIC_setDrTxpow(DR_SF12, 14); // or other SF you want`
but no success. Still it is sending with SF9 as before.

slavendam, 2023-10-30:
> I am using this example with ABP method, but I am facing same problem.

ABP doesn't go through process of joining so EV_JOINED is never called.
For ABP you need to set this at the end of void setup() function.
On line 266 in your example you already have function that setup SF. Just change value.
https://github.com/martinius96/hladinomer-studna-scripty/blob/0a5983aaaee78427656a8620987ae33e96245f02/examples/Hladinomer/LoRaWAN/LoRaWAN.ino#L266

martinius96, 2023-11-07:
still not solved by me