mcci-catena/arduino-lmic #944: Newbie question - transmitting small packets at near real time speeds

yaydogan, 2023-07-16:
Hi,

I am new to Lora  network and want to understand the feasibility of what I am trying to build. I am planning to build a sensor with a GPS and want to send location information every 500msec or less. The payload will be small - may be 12 bytes. Sensor will be always powered so I am not worried about power consumption. The sensor will not receive any data from network so there won't be any traffic on downlink.  Is this possible with LMIC library?

Thanks in advance.

terrillmoore, 2023-07-17:
Not possible as stated. The LMIC library waits for the RX1 window to close unconditionally. That's network specific, but at least 1 second, and 5 seconds on The Things Network.

However, if latency is not an issue, you can aggregate location readings, and transmit a large record. At SF7 or SF8, you can transmit 240 bytes. Location fixes are typically 4 bytes of data if you send in binary, so 240 bytes is 60 fixes; if you send every 30 seconds, you'll get the required data density (but there will be latency for the older readings).

simortus, 2023-10-05:
Another thing is your device will violate the free band restriction, which is 1% OAT (on air time) for end-nodes.