mcci-catena/arduino-lmic #951: no JoinAccept

danialkoosha56, 2023-12-27:
Hi
i have problem. every details is written below:

- Version of Arduino : Platform IO
- Network provider : The Things Network
- Region : EU868
- Board : esp32 wroom
- Radio : RFM96 (HopeRF, SX1276)

Im using "ttn-otaa.ino" example without any changes

besed on ttn console i have this messages:
"forward join_accept message"
"Successfully processed join_request"
"accespt join_request message
[image]
[image]
"

and this is my serial Log:
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Starting
RXMODE_RSSI
3827: engineUpdate, opmode=0x8
Packet queued
3856: EV_JOINING
3863: engineUpdate, opmode=0xc
198902: engineUpdate, opmode=0xc
198926: EV_TXSTART
199011: TXMODE, freq=868500000, len=23, SF=7, BW=125, CR=4/5, IH=0
499005: setupRx1 txrxFlags 00 --> 01
start single rx: now-rxtime: 5
499638: RXMODE_SINGLE, freq=868500000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 531463 rxtime: 499630 entry-rxtime: 31833 now-entry: 5 rxtime-txend: 296754
558382: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 4
559013: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 610230 rxtime: 559006 entry-rxtime: 51224 now-entry: 6 rxtime-txend: 356130
610253: processRx2Jacc txrxFlags 0x2 --> 00
610303: EV_JOIN_TXCOMPLETE: no JoinAccept
610953: engineUpdate, opmode=0xc
4418101: engineUpdate, opmode=0xc
4418123: EV_TXSTART
4418204: TXMODE, freq=868300000, len=23, SF=7, BW=125, CR=4/5, IH=0
4718198: setupRx1 txrxFlags 00 --> 01
start single rx: now-rxtime: 5
4718831: RXMODE_SINGLE, freq=868300000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 4750655 rxtime: 4718823 entry-rxtime: 31832 now-entry: 6 rxtime-txend: 296754
4777574: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 4
4778206: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 4829423 rxtime: 4778199 entry-rxtime: 51224 now-entry: 6 rxtime-txend: 356130
4829445: processRx2Jacc txrxFlags 0x2 --> 00
4829512: EV_JOIN_TXCOMPLETE: no JoinAccept
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

and i check DIO1 pin with logic analyzer it goes high and creat intrrupt 2 times

HilmarSt, 2025-01-30:
I've the same problem - but it regularely disappears after ~10 Minutes (~600 seconds):

```
...
330 s: EV_JOIN_TXCOMPLETE: no JoinAccept
438 s: EV_TXSTART
444 s: EV_JOIN_TXCOMPLETE: no JoinAccept
553 s: EV_TXSTART
558 s: EV_JOINED
558 s: EV_TXSTART
563 s: EV_TXCOMPLETE (includes waiting for RX windows)
578 s: EV_TXSTART
583 s: EV_TXCOMPLETE (includes waiting for RX windows)
...
```

terrillmoore, 2025-01-31:
In EU868, this means one of the following:

1. you are not receiving messages. If you have two devices, try the raw LoRa example. 
2. your gateway is receiving but not transmitting, due to a failure in the gateway, either software or hardware.
3. your gateway is correctly not transmitting because otherwise it would exceed the regional duty cycle limit. 

In any case, this is extremely unlikely to be a bug in the LMIC per se, so I will convert this to a discussion. We can change it back to an issue if I'm mistaken.