mcci-catena/arduino-lmic #963: 30 minutes delay between LMIC_setTxData2 and actual send

i-g-g-y, 2024-07-25:
Hello.

There is a serious bug in the timing of the library.

I am using;
- version 4.1.1
- IDE 2.3.2
- network is Helium
- EU868
- nRF52840 custom board
- nRF52840
- RFM95

I keep track of time and send out a packet every 2 hours with;

`
    // Check if there is not a current TX/RX job running
    if (LMIC.opmode & OP_TXRXPEND) {
    Serial.println(F("OP_TXRXPEND, not sending"));
    } else {
    // Prepare upstream data transmission at the next possible time.
    last_tx_time = millis();
    Serial.print("\r\nTx req.");
    os_setTimedCallback(&sendjob, os_getTime()+ms2osticks(100), do_send);

      LMIC_setTxData2(1, txBuffer, sizeof(txBuffer)-1, 0);

      Serial.println(F("Packet queued!"));
  }
`

**Expected behavior**

This should send data out after 100ms, as the os_runloop_once() is constantly being called.

Most times it works flawlessly, but sometimes I get 30 minutes of delay between queuing the packet and the actual send. Why is that?

23:02:53.432 -> Tx req.Packet queued!
23:02:53.571 -> Packet queued!
23:03:26.946 -> ........................1341662618: EV_TXSTART
23:26:46.086 -> 1341829732: EV_TXCOMPLETE (includes waiting for RX windows)

qiweimao, 2025-02-20:
Hi, did you resove this issue?

i-g-g-y, 2025-04-27:
I modified the code to update LMIC time more often, namely every second.