mcci-catena/arduino-lmic #994: Connecting to AWS IotCore LNS

jshakar, 2025-01-05:
I am trying to connect a LilyGo T-Beam to a Raspberry Pi running basic station on AWS IOT Core LNS. I have the basic station part working, 

I am not sure if the T-beam (Node) running the MMCI library is working correctly. 

Environment:
- Version of LMIC being used: MCCI LoRaWAN LMIC library@^4.1.1
- Version of Arduino IDE being used: Platform IO
- Network provider: AWS IOT Core
- Region US915
- Board: LilyGo T-Beam
- Radio: SX1276

[image]

Here is the Router Config from basics station if it may be helpful:
```
2025-01-05 00:21:41.476 [AIO:XDEB] [4|WS] < {"msgtype":"router_config","NetID":null,"JoinEUI":null,"region":"US902","hwspec":"sx1301/1","freq_range":[902000000,928000000],"DRs":[[10,125,0],[9,125,0],[8,125,0],[7,125,0],[8,500,0],[-1,0,0],[-1,0,0],[-1,0,0],[12,500,1],[11,500,1],[10,500,1],[9,500,1],[8,500,0],[7,500,1],[-1,0,0],[-1,0,0]],"sx1301_conf":[{"radio_0":{"enable":true,"freq":902700000},"radio_1":{"enable":true,"freq":903700000},"chan_FSK":{"enable":false},"chan_Lora_std":{"enable":true,"radio"
2025-01-05 00:21:41.476 [AIO:XDEB] [4|WS] . :0,"if":300000,"bandwidth":500000,"spread_factor":8},"chan_multiSF_0":{"enable":true,"radio":0,"if":-400000},"chan_multiSF_1":{"enable":true,"radio":0,"if":-200000},"chan_multiSF_2":{"enable":true,"radio":0,"if":0},"chan_multiSF_3":{"enable":true,"radio":0,"if":200000},"chan_multiSF_4":{"enable":true,"radio":0,"if":400000},"chan_multiSF_5":{"enable":true,"radio":1,"if":-400000},"chan_multiSF_6":{"enable":true,"radio":1,"if":-200000},"chan_multiSF_7":{"enable
2025-01-05 00:21:41.476 [AIO:XDEB] [4|WS] . ":true,"radio":1,"if":0}}],"protocol":0,"regionid":0}

```

I also have a support thread going on AWS IQ: https://repost.aws/questions/QUsqYcZCQURlq5xFqm9myXaA/aws-iot-lorawan-unable-to-connect-node-to-gateway

jshakar, 2025-01-06:
It seems the LMIC library is outputting on sub-band 2, even though sub-band 1 is specified in code. Not sure how to get the node to output using sub-band 1. 

I have done the following in the LMIC setup :

`LMIC_selectSubBand(1);`

As well as the following in the platformio.ini:

```
build_flags = 
    -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS
    -D CFG_us915=1
    -D CFG_sx1276_radio=1
    -D LMIC_SELECT_SUBBAND=1  ; This selects sub-band 1 

```

terrillmoore, 2025-01-06:
Sorry you're having problems.

1. There is confusion on naming of subbands.  Some people use origin 0, other people use origin 1. The only way to disambiguate is to mention the actual channels used. The LMIC uses origin 0, so LMIC subband 1 is channels 8~15/65, which is what some other people call subband 2. (In fact, the LoRa Alliance ultimately settled on this, but it was long after the LMIC code decided to use zero origin.)

2. Selecting a subband when you're doing OTAA is really obsolete and bad practice in 2025. The LMIC will send join requests on all subbands until it gets a response. The LNS will then tell the LMIC what channels to use.

Best regards,
--Terry

jshakar, 2025-01-06:
Hi Terry, thanks for the response. Your comment 1 fixed the sub-band mismatch I was having. Your comment 2 makes sense (although I don't know how to implement that right now and will save that for another time. Defining the sub-bands explicitly now that they match seems to be okay for testing purposes)

I am now getting a ping showing in the gateway logs. However the pings are few far between. Like 10-20 minutes between logs. Here is what the log looks like from the basic station:

Looks to be the CRC is failing. and the TX power of the node is very low?

(notes)
I've set the tx power in the code explicitly:
```
// Set Data Rate and TX Power
  LMIC.txpow = 21; // Set TX power to 21 dBm (max for US915)
```
and for the LMIC to listen on SF8:

```
 // Ensure the node expects SF8 for RX2 downlinks
  LMIC.dn2Dr = DR_SF8;
```

```

2025-01-06 19:31:05.896 [SYN:XDEB] SYNC: ustime=0x00002BF2B230 (Q=181): xticks=0x26ca8223 xtime=0xF5000026CA8223 - PPS: pps_xticks=0x00000000 (0) pps_xtime=0x0 (pps_en=0)
2025-01-06 19:31:07.146 [HAL:XDEB] [rx_buffer_fetch:154] -----------------
2025-01-06 19:31:07.146 [HAL:XDEB] [rx_buffer_fetch:155] rx_buffer_fetch: nb_bytes to be fetched: 23 (23 0)
2025-01-06 19:31:07.146 [HAL:XDEB] [rx_buffer_fetch:165] RX BUFFER: A5C00007500D00270C00E67D796345FC6BB1DB00000073 
2025-01-06 19:31:07.146 [HAL:XDEB] [rx_buffer_fetch:177] INFO: syncword found at idx 0
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:244] INFO: pkt syncword found at index 0
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:275] Packet checksum OK (0x73)
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:311] -----------------
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:312]   modem:      13
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:313]   chan:       7
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:314]   size:       0
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:315]   crc_en:     0
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:316]   crc_err:    0
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:317]   sync_err:   0
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:318]   hdr_err:    0
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:319]   timing_set: 0
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:320]   codr:       0
2025-01-06 19:31:07.147 [HAL:XDEB] [rx_buffer_pop:321]   datr:       5
2025-01-06 19:31:07.148 [HAL:XDEB] [rx_buffer_pop:322]   num_ts:     0
2025-01-06 19:31:07.148 [HAL:XDEB] [rx_buffer_pop:331] -----------------
2025-01-06 19:31:07.148 [HAL:XDEB] [sx1302_parse:1946] [7 0x11]
2025-01-06 19:31:07.148 [HAL:XDEB] [sx1302_parse:1959] Note: LoRa packet (modem 13 chan 7)
2025-01-06 19:31:07.148 [HAL:ERRO] [timestamp_counter_correction:463] wrong coding rate (0)
2025-01-06 19:31:07.148 [HAL:XDEB] [sx1302_rssi_get_temperature_offset:2283] INFO: RSSI temperature compensation:
2025-01-06 19:31:07.148 [HAL:XDEB] [sx1302_rssi_get_temperature_offset:2284]        coeff_a: 0.000
2025-01-06 19:31:07.148 [HAL:XDEB] [sx1302_rssi_get_temperature_offset:2285]        coeff_b: 0.000
2025-01-06 19:31:07.148 [HAL:XDEB] [sx1302_rssi_get_temperature_offset:2286]        coeff_c: 20.410
2025-01-06 19:31:07.148 [HAL:XDEB] [sx1302_rssi_get_temperature_offset:2287]        coeff_d: 2162.560
2025-01-06 19:31:07.148 [HAL:XDEB] [sx1302_rssi_get_temperature_offset:2288]        coeff_e: 0.000
2025-01-06 19:31:07.148 [HAL:XDEB] [lgw_receive:1320] INFO: RSSI temperature offset applied: 1.375 dB (current temperature 32.0 C)
2025-01-06 19:31:07.148 [HAL:XDEB] [lgw_receive:1323] INFO: nb pkt found:1 left:0
2025-01-06 19:31:07.148 [HAL:XDEB] [lgw_receive:1337]  --- OUT
2025-01-06 19:31:07.148 [RAL:DEBU] [CRC FAIL] 903.700MHz -6.50/-93.0 SF?? (mod=16/dr=5/bw=4) xtick=26dd8b5f (652053343) 0 bytes: 

```

Here is the second ping i've gotten thus far:

```
2025-01-06 20:03:11.920 [SYN:XDEB] SYNC: ustime=0x00009EBF789E (Q=185): xticks=0x99974c36 xtime=0xF5000099974C36 - PPS: pps_xticks=0x00000000 (0) pps_xtime=0x0 (pps_en=0)
2025-01-06 20:03:12.807 [HAL:XDEB] [rx_buffer_fetch:154] -----------------
2025-01-06 20:03:12.807 [HAL:XDEB] [rx_buffer_fetch:155] rx_buffer_fetch: nb_bytes to be fetched: 67 (67 0)
2025-01-06 20:03:12.807 [HAL:XDEB] [rx_buffer_fetch:165] RX BUFFER: A5C02C03560800790F0F316B6C68688C78A426DD0EF4E4228D74174A8037BB996F3B8746CCFAFB7BFAEC0AEA106A235F0F4FB8CB5A00E46861F7FB6400973400000013 
2025-01-06 20:03:12.807 [HAL:XDEB] [rx_buffer_fetch:177] INFO: syncword found at idx 0
2025-01-06 20:03:12.808 [HAL:XDEB] [rx_buffer_pop:244] INFO: pkt syncword found at index 0
2025-01-06 20:03:12.808 [HAL:XDEB] [rx_buffer_pop:275] Packet checksum OK (0x13)
2025-01-06 20:03:12.808 [HAL:XDEB] [rx_buffer_pop:311] -----------------
2025-01-06 20:03:12.808 [HAL:XDEB] [rx_buffer_pop:312]   modem:      8
2025-01-06 20:03:12.808 [HAL:XDEB] [rx_buffer_pop:313]   chan:       3
[... 4 lines cut ...]
2025-01-06 20:03:12.808 [HAL:XDEB] [rx_buffer_pop:318]   hdr_err:    0
2025-01-06 20:03:12.808 [HAL:XDEB] [rx_buffer_pop:319]   timing_set: 0
2025-01-06 20:03:12.808 [HAL:XDEB] [rx_buffer_pop:320]   codr:       3
2025-01-06 20:03:12.808 [HAL:XDEB] [rx_buffer_pop:321]   datr:       5
2025-01-06 20:03:12.808 [HAL:XDEB] [rx_buffer_pop:322]   num_ts:     0
2025-01-06 20:03:12.809 [HAL:XDEB] [rx_buffer_pop:331] -----------------
2025-01-06 20:03:12.809 [HAL:XDEB] [sx1302_parse:1946] [3 0x11]
2025-01-06 20:03:12.809 [HAL:XDEB] [sx1302_parse:1959] Note: LoRa packet (modem 8 chan 3)
[... cut ...]
2025-01-06 20:03:12.809 [HAL:XDEB] [lgw_receive:1320] INFO: RSSI temperature offset applied: 1.560 dB (current temperature 35.4 C)
2025-01-06 20:03:12.809 [HAL:XDEB] [lgw_receive:1323] INFO: nb pkt found:1 left:0
2025-01-06 20:03:12.809 [HAL:XDEB] [lgw_receive:1337]  --- OUT
2025-01-06 20:03:12.809 [RAL:DEBU] [CRC FAIL] 902.900MHz -7.00/-116.8 SF?? (mod=16/dr=5/bw=4) xtick=99a4b780 (2577708928) 44 bytes: 0F316B6C68688C78A426DD0EF4E4228D74174A8037BB996F3B8746CCFAFB7BFAEC0AEA106A235F0F4FB8CB5A
```

jshakar, 2025-01-06:
UPDATE:

Seems the gateway was picking up some other lora signal. Hence why the receive was weak, few and far between, and had a changing packet size.

The node is sending a join accept (len = 23) every ten seconds. The gateway is not picking this up. 

At least I know the gateway does seem t9o pick up RF signal, leading me to think something is wrong with the node.