mcci-catena/arduino-lmic #1001: NUCLEO-WL55JC1 with STM32Duino and SX1272: Crashing at os_init();

SeppeBudenaers, 2025-03-04:
Hi everyone,  

I'm trying to run a NUCLEO-WL55JC1 with STM32Duino and an SX1272 module. I set a custom pin mapping as explained in the documentation and updated the radio configuration in the config file. However, the system crashes when entering `os_init();`.  
```CMD
Starting
FAILURE 
c:\Users\seppe\Documents\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\lmic\oslmic.c:53
```
I suspect the issue might be related to SPI pin definitions (`MOSI`, `MISO`, `CLK`), as I haven't explicitly set them and I'm unsure which pins are being used.  

Does anyone know how and where to define these SPI pins, or could the issue be caused by something else?  

Any help would be greatly appreciated!  

---
```CPP
/*******************************************************************************
 * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
 * Copyright (c) 2018 Terry Moore, MCCI
 *
 * Permission is hereby granted, free of charge, to anyone
 * obtaining a copy of this document and accompanying files,
 * to do whatever they want with them without any restriction,
 * including, but not limited to, copying, modification and redistribution.
 * NO WARRANTY OF ANY KIND IS PROVIDED.
 *
 * This example sends a valid LoRaWAN packet with payload "Hello,
 * world!", using frequency and encryption settings matching those of
 * the The Things Network.
 *
 * This uses OTAA (Over-the-air activation), where where a DevEUI and
 * application key is configured, which are used in an over-the-air
 * activation procedure where a DevAddr and session keys are
 * assigned/generated for use with all further communication.
 *
 * Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in
 * g1, 0.1% in g2), but not the TTN fair usage policy (which is probably
 * violated by this sketch when left running for longer)!

 * To use this sketch, first register your application and device with
 * the things network, to set or generate an AppEUI, DevEUI and AppKey.
 * Multiple devices can use the same AppEUI, but each device has its own
 * DevEUI and AppKey.
 *
 * Do not forget to define the radio type correctly in
 * arduino-lmic/project_config/lmic_project_config.h or from your BOARDS.txt.
 *
 *******************************************************************************/

#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

//
// For normal use, we require that you edit the sketch to replace FILLMEIN
// with values assigned by the TTN console. However, for regression tests,
// we want to be able to compile these scripts. The regression tests define
// COMPILE_REGRESSION_TEST, and in that case we define FILLMEIN to a non-
// working but innocuous value.
//
#ifdef COMPILE_REGRESSION_TEST
# define FILLMEIN 0
#else
# warning "You must replace the values marked FILLMEIN with real values from the TTN control panel!"
# define FILLMEIN //(#dont edit this, edit the lines that use FILLMEIN)
#endif

// This EUI must be in little-endian format, so least-significant-byte
// first. When copying an EUI from ttnctl output, this means to reverse
// the bytes. For TTN issued EUIs the last bytes should be 0xD5, 0xB3,
[... 181 lines cut ...]

    #ifdef VCC_ENABLE
    // For Pinoccio Scout boards
    pinMode(VCC_ENABLE, OUTPUT);
    digitalWrite(VCC_ENABLE, HIGH);
    delay(1000);
    #endif

    // LMIC init
    os_init();
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();
    //SetclockError added to fix joining timing issue 
    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100); 

    // Start job (sending automatically starts OTAA too)
    do_send(&sendjob);
   // Data rate can be defined here.if not defined a rate is assigned
    //LMIC_setDrTxpow (DR_SF9, 14);
}

void loop() {
    os_runloop_once();
}
```

terrillmoore, 2025-03-04:
Looking at the code, this is a failure because os_init_ex() returned 0. This can only happen because radio_init() returned 0. This happens in turn because the version number doesn't match expectations.

There are two reasons this might happen:

1. Mismatch between the `CFG_..._radio` definition and your hardware
2. A new version number that Semtech has added

I've seen both problems. The thing to do is edit the code and print out at line 1110 of radio.c what version number you're getting back.

SeppeBudenaers, 2025-03-04:
I’ll look into it further once I have access to the hardware again.  

Thank you for the quick reply!

SeppeBudenaers, 2025-03-07:
@terrillmoore I’m sorry, but I don’t see a `Radio.c` file anywhere.  

[image]  

You have these two calls that lead to `Radio_Init()`:  

[image]  

However, `Radio_Init()` is only defined in `oslmic.h`. It is called in the function shown above, but there is no actual implementation of the function anywhere.  

[image]  

 > As far as I can see