mcci-catena/arduino-lmic #1006: Multiple node instances to talk to different networks

i-g-g-y, 2025-04-27:
I would like to use the same hardware to talk to multiple different LoRaWAN networks, for instance TTN and Helium.

This would allow to improve sensor availability.

I would like to have multiple instances completely independent of each other but running on the same hardware at the same time.

terrillmoore, 2025-04-28:
Thanks for your suggestion.

The code is unfortunately very non-re-entrant. However, you could fake this out by initializing for network A, doing your work with network, A, waiting for the LMIC to go idle, then saving the state for network A, restoring the state for network B, operating on network B, etc. Won't work at all for things like class B or class C, but for class A,  The arduino-lorawan library has code that shows how to save / restore state.

Any thing more sophisticated will require a major rewrite.

Best regards,
--Terry

i-g-g-y, 2025-04-28:
Hello Mr Moore for considerong this feature and taking the time to evaluate it. I will try your suggestion, as my nodes are all class A. 

Thank you and best regards,
Igor