mcci-catena/arduino-lmic #1009: Invalid condition when checking the chip mode?

Maddocrus, 2025-06-02:
**Describe the bug**
Hello plz check this:
..\MCCI LoRaWAN LMIC library\src\lmic\radio_sx126x.c
line 209 :
```
// ----------------------------------------
// GetStatus        **  Chapter 13 GetStatus
#define SX126x_GETSTATUS_CHIPMODE_MASK 0x07 // Bits 6:4
#define SX126x_CHIPMODE_STDBY_RC       0x02
#define SX126x_CHIPMODE_STDBY_XOSC     0x03
#define SX126x_CHIPMODE_FS             0x04
#define SX126x_CHIPMODE_RX             0x05
#define SX126x_CHIPMODE_TX             0x06

```

..\MCCI LoRaWAN LMIC library\src\lmic\radio_sx126x.c
**Problem here that will always TRUE->**
Line 1160:
```
   // Sleep needs to be entered from standby_RC mode 
     if ((getStatus() | SX126x_GETSTATUS_CHIPMODE_MASK) != SX126x_CHIPMODE_STDBY_RC) {
        setStandby(STDBY_RC);
    }
    setSleep(0);
    return 1;

```

**Environment**

- CPU STM32f103
- Radio SX1262 

**Expected behavior**

Normally, the response is 0x22 and I expect that the code will not turn on the one that is already active every time.

**Screenshots**

[image]

**Additional context**

This condition occurs several times.

Maddocrus, 2025-06-03:
I have corrected to the option you suggested and shifted the macro definitions 

```
// ----------------------------------------
// GetStatus        **  Chapter 13 GetStatus
#define SX126x_GETSTATUS_CHIPMODE_MASK 0x70 // Bits 6:4
#define SX126x_CHIPMODE_STDBY_RC       0x20
#define SX126x_CHIPMODE_STDBY_XOSC     0x30
#define SX126x_CHIPMODE_FS             0x40
#define SX126x_CHIPMODE_RX             0x50
#define SX126x_CHIPMODE_TX             0x60
   
if ((getStatus() & SX126x_GETSTATUS_CHIPMODE_MASK) != SX126x_CHIPMODE_STDBY_RC) {

```
since there is a comment on GetStatus:
..\MCCI LoRaWAN LMIC library\src\lmic\radio_sx126x.c
Line 733:
```
// Chapter 13.5: Communication Status Information

// Chip mode is (getStatus | 0x70)
static u1_t getStatus(void) {
    u1_t status;
    lmic_hal_spi_read_sx126x(GetStatus, NULL, 0, &status, 1);
    return status;
}
```

terrillmoore, 2026-10-07:
Fixed in #1110 (merged as c61815c, 6.1.0-pre9): the chip-mode test now masks bits 6:4 and compares with AND (`isChipModeStdbyRc()`), and the mask is 0x70. Thanks to @Maddocrus for the report and analysis.

closed 2026-10-07 by terrillmoore